xarial/cad-plus #15: PDF export is corrupt\zero size

ntzallas, 2020-08-05:
Running:
eDrawings x64 edition 2020 SP 3.0 (28.3.0.0025)
PDM Professional 2020 SP2.0
Adobe acrobat DC 2020.009.20074

After exporting slddrw to pdf, opening pdf in adobe acrobat DC results in below error
Have tried multiple times, same resulting error

Only notable difference compared to xPort video tutorial demo is that the source path of slddrw files is in a local PDM Pro vault view 

Exported pdf files are zero size. I cannot even attach them to this ticket for reference

Hope this helps

[image]

[image]

[image]

[image]

artem1t, 2020-08-25:
Could you please make sure that those drawings are cached locally by PDM.

ntzallas, 2020-08-25:
> Could you please make sure that those drawings are cached locally by PDM.

[image]

[image]

Export now says completed successfully, however there is no files in the destination folder

I should mention that our company has now recently upgraded us to 2020 SP 4.0 (28.4.0.0015). In case that is the culprit

artem1t, 2020-08-25:
Are you on Windows 10? Could you please also try to export outside of PDM vault - just want to see if this is PDM related issue.

ntzallas, 2020-08-25:
Same result, no export file

[image]

[image]

[image]

[image]

artem1t, 2020-08-27:
Tool is using 'Microsoft Print To PDF' printer to export PDFs. Could you please try to manually export PDF from eDrawings via this printer, does it work?

ntzallas, 2020-08-27:
Using "Microsoft Print to PDF" works correctly. PDF file is created and is readable

artem1t, 2020-08-27:
I have found an open SPR in the SW Knowledge Base regarding the corrupted PDF files with Print5 method. I believe this is a problem. However, it is still working on my machine (even when I upgraded to SP4). But I can reproduce this issue on a brand new machine with eDrawing 2020 SP4. Unfortunately seems like it is not possible to download previous versions of eDrawings to validate the regression.

ntzallas, 2020-08-27:
I might be misunderstanding, so please ignore if this is redundant. 

Is not possible simply to download the SP 2.0 or 3.0 pack from the below location and just check "edrawings" during installation, and nothing else?  

Either way, glad you were able to find a possible issue

[image]

artem1t, 2020-08-27:
It might work as well. But it used to be on 'Free Tools tab', now you can only download latest version there. To download older it is required to fill the form here: https://www.solidworks.com/support/free-product-form. I will contact my VAR to get the download - just cannot do it immediately and it may take some time.

rikkis88, 2021-12-01:
Im having this same issue as well. Was there a solution? Im using Solidworks 2021 Version, we dont have PDM