xarial/cad-plus #41: custom toolbar rename swp issue

EddyAlleman, 2020-10-21:
When you rename the macro .swp file in windows explorer the dialog doesn’t show a warning that the file doesn’t exist anymore and when you try to make adjustments in the dialog, you get a crash whatsoever.

I could only solve this by renaming to the previous filename of the swp file in windows explorer.

artem1t, 2020-10-25:
Errors now marked in red and tooltip display the issues:

[image]

EddyAlleman, 2020-10-25:
Hi Artem,

 

Very distinctive.! 

 

I Will install the latest version using the App center link you sent me.

 

Eddy

 

 <https://user-images.githubusercontent.com/44194903/97106527-518f5780-1716-11eb-85c8-e7991dd763b3.png>

closed 2020-10-25 by EddyAlleman