xarial/cad-plus #62: "New version is available" popup keeps appearing

EddyAlleman, 2020-11-18:
The following popup keeps appearing
[image]

But I have this test version installed:
0.5.1286.4
[image]

thanks,

Eddy

artem1t, 2020-11-20:
Hi Eddie,

This popup will be shown when new production version is available. Please note that test package of the same version is upgradable to the production package of the same version as well as older production version is upgradabel to the newer production version. So in your case you can install the production version from here and the popup will go away: https://install.appcenter.ms/orgs/xarial/apps/cad-toolset/distribution_groups/production

Thanks,
Artem

EddyAlleman, 2020-11-20:
Ok thanks for the clarification.

Eddy

closed 2020-11-20 by EddyAlleman