xarial/cad-plus #81: Add ability to export all underlying drawings from Assembly to PDF

Quasardian, 2021-02-01:
We're looking to digitizing our printed SolidWorks drawings to PDF's.
For this we need a Macro, which to my knowledge does not exist yet.

I've seen multiple PDF export macro's but none of which would fit our needs.

We have many different drawings in underlying parts and assembly's in a main assembly, which we would like to export all at the same time.
But since this different for each project we would also need to be able to select specified sheets and a folder.
I put an example in the attachment, which I made with Excel and Paint.
I tried to optimize the layout as best I could, but I'm open to suggestions.

I hope someone can help me with this, thank you in advance!

[image]

Quasardian, 2021-02-01:
(Moved from Codestack to CAD+)

artem1t, 2021-02-01:
It is currently planned to be implemented in next release (part of the Community edition). For now it will find referenced drawing in the specified folder(s). Once the #31 is implemented, this will also extract drawing sheets.

Quasardian, 2021-04-12:
I was wondering if the CAD+ project still active is?

artem1t, 2021-04-13:
Yes, in fact, I am rolling the private beta from today. Here is the video which covers this functionality: [How to print and export all drawings used by components in the SOLIDWORKS assembly](https://www.youtube.com/watch?v=5akBj5gd68Q). If you want an access to private beta, please send me an email to beta@xarial.com