xarial/cad-plus #85: Add the ability to execute scripts like Python (or C#)

Glutenberg, 2021-06-25:
Add the ability to run out of process scripts like Python (or C#). 

It should work as follows: 
1. Load and activate a SolidWorks model
2. Execute the selected Python (or C#) 
3. Wait for Script to Finish
4. Move on to the next model

This will allow python scripts written using the win32 package to be used which will make it easier to:
1. Interact with APIs
2. Interact with DBs
3. Manipulate Data 

At present you can write a wrapper script in VBA to execute the python script but then this has to be done for every single script. 

For those who may care see code below: 

Dim swApp As Object

Sub main()

'Variable Declarations
Dim Shell As Object
Dim PythonExe As String
Dim PythonScript As String
Dim WaitOnReturn As Boolean: WaitOnReturn = True
Dim WindowStyle As Integer: WindowStyle = 1

PythonExe = """C:\Windows\pyw.exe"""
PythonScript = """C:\Path\To\helloworld.pyw"""

Set swApp = Application.SldWorks
Set Shell = VBA.CreateObject("WScript.Shell")

'Shell.Run (strCommand, [intWindowStyle], [bWaitOnReturn])
Shell.Run PythonExe & PythonScript, WindowStyle, WaitOnReturn

End Sub

artem1t, 2021-06-28:
@Glutenberg , thanks a lot for the idea. Next version (currently in the dev branch) will support C#/VB.NET based modules (xCAD Macros) both in Batch+ and Toolbar+. This will be run out of the process for Batch+ stand-alone and in-process for Batch+ in assemblies and Toolbar+. Do you have some example of a simple python script (hello world will do) which is utilizing SOLIDWORKS API? Unfortunately, I have no experience in Python, but seen several requests before regarding Python and SW API so keen to look into this request closer and enable the proper support.

Glutenberg, 2021-07-09:
@artem1t See short example script below for changing a custom property. Full disclosure, swtoolkit is my pet project  (which is essentially just a wrapper) with only a few users. However, Solidworks can be automated by using the more popular win32 and pythoncom packages 

`
from swtoolkit import SolidWorks

def main():
    sldworks = SolidWorks()
    model = sldworks.get_model()
    prop_manager = model.extension.custom_property_manager("Default")

    prop_manager.add(
         "DESCRIPTION",
         "TEXT",
         "ARC REACTOR",
         overwrite_existing="DELETE AND ADD",
     )
   
if __name__ == "__main__":
    main()

`

The python script runs out of process. Essentially what I'm asking for is for batch+ to be able to run the python script. As I mentioned in my first post I currently write a vba script to execute the python script which just gets messy and requires diligent "book keeping" when changes are made and files are moved.