xarial/cad-plus #94: Possibility to open Excel file after creating it

EddyAlleman, 2021-08-07:
Would be nice if you add an option to do this or not in settings. 
This could be a checkmark to always open it after saving so you can check the result immediately.
Another advantage is users don’t have to search for it in windows explorer anymore.
BOM+

artem1t, 2021-08-16:
Ideally, this is to be implemented as the checkbox in the Save As dialog itself. There is no trivial way to do this, but still worth researching.

EddyAlleman, 2021-08-17:
To make it easier to implement, this could also be in a preferences dialog box separately.
You mostly want it always to show up or never to show up. And it is not a big deal if you have t open it manually afterwards anyway.