xarial/cad-plus #101: Feature request: possibility to debug into macro when running Batch+

EddyAlleman, 2021-08-16:
We can debug our macro when our solidworks files is open and just run the macro.

This means that the API takes into account that manually files where opened.
If we open files through code some Solidworks settings are not change like the working directory.

So testing in the Batch+ environment, would find those issues.

Remark: it is possible to use  MsgBox in your code and use the "Silent" setting:
[image]

Which will show the messages, but closes them automatically

Journal still shows them nevertheless:
[image]

artem1t, 2021-08-17:
You can add the following line in your macro and it will break so you can debug your macro as normal.

~~~ vb
Debug.Assert False
~~~

I have added the 'Documentation label so it will be added to the docs

EddyAlleman, 2021-08-17:
Thanks a lot Artem

Works great!

closed 2021-08-17 by EddyAlleman