xarial/cad-plus #134: Batch+ Stand-Alone - Wrong dependencies?

Quasardian, 2021-12-07:
I see a potential problem exporting drawings.
As an example I'll use the screenshot from below.
Under my main assembly there is a file called T1694, and under it is a file called Z654584.
Since file Z654584 is in the list, it will now list every assembly where it is used.
For this example it's only 6 different drawings, but some parts can list 100+ drawings.

[image]

Top-level dependencies would not work for me, we would need second level or even third level dependencies in some cases.

artem1t, 2021-12-07:
Batch+ is looking for top-level references from the drawings to decide if the specific file belongs to a drawing. For example here is the assembly (similar to your structure) [Refs.zip](https://github.com/xarial/cad-plus/files/7666947/Refs.zip)

Here is the BOM of this assembly

[image]

And this is how it is looking for references:

[image]

* Part1 is used in both Part1 drawing and SubAssem1 drawing (another drawing view)
* Part3 is only used in Part3 drawing and only this drawing is shown despite Part3 is also a part of SubAssem1 and SubAssem2 but these drawings are not included (correct behavior, and I believe this is what you expect from Z6545584)
* Part2 is only used directly in the Part2 drawing, however, it is also added to the view palette of the SubAssem3 drawing and thus makes it top-level reference

[image]

Could you please display the references (SOLIDWORKS menu File-Find References) for T1603, I believe you will have Z654584 listed there as the top-level reference either because there is a separate view for this part or it is listed in the view palette

Quasardian, 2021-12-07:
I think I understand what you're saying, here is the reference screenshot that you asked for.
yes Z654584 is listed as a part in T1603 for sure.

[image]

But T1603 is not part of my TopAssembly.

So the reference would look like:
[image]

But it would still show up like this:
[image]

Because it somehow finds another reference in PDM "Where used"?